RobTillaart/MCP23S17 #25: Could I contact you with a help request for once?

KarenColumbo, 2023-08-25:
**Dear Rob, I've been falling into despair with an attempt to control FSYNC pins of AD9833 boards via an MCP23S17.** 

PJRC forum isn't much help because not many users there are fluent with the MCP23S17. Plus: My problem may be pretty n00b, so it isn't too interesting. 

If you're not interested, please just drop it here and delete this "issue". I just don't know how else to contact you, that's why I write here.

The main problem, I think, is that AD9833 need SPI_MODE2, while the MCP23S17 is content with SPI_MODE0. 

I'd be ready to send you some AD9833 boards (or pay for them on Amazon) if you could find the motivation and some of your valulable time to help an analog synthesizer freak with this.

I have a simple working example code where I write the AD9833 directly from the Teensy 4.1:
https://github.com/KarenColumbo/AD9833-DCO8/blob/main/src/main.cpp

There's some code of what i want to do, but it doesn't work:
https://github.com/KarenColumbo/AD9833-DCO8/blob/main/test/AD9833_Repo/MCPToggle-WithLibrary-Tillaart.cpp

Goal is to control the FSYNC pins of up to 16 AD9833 boards. I'm buffering all 4 SPI line between Teensy 4.1 and the rest of the chips with an 74HC125 buffer.

I'd, of course, be willing to further sponsor your great work here on Github. it's the least i can do.

If you use E-Mail, I'm at office@mrscolumbo.com

Thank you for reading this and for your work for the community!

KarenColumbo, 2023-08-25:
If you're willing to help me (if there's a solution at all), I'll gladly put your name somewhere prominent on the really good synth module I'm trying to make here.

RobTillaart, 2023-08-25:
Hi Andreas,

Interesting project!

I'm quite busy on a few other issues and projects so I can only give some quick first thoughts.
Q1: what processor do you use?
Q2: do you have a few free pins?

My first thought is to change the hardware SPI for a software SPI for the MCP23S17. 
It will teach us two things
- proof of concept, does it work?
- if so, is it fast enough for your project?

RobTillaart, 2023-08-25:
@KarenColumbo 
BTW thanks for your recent support!

RobTillaart, 2023-08-25:
I accepted the invitation and had a quick look at the two links of the AD9833-DCO8 repo.
Furthermore I created an issue to have a place to discuss the repo.

RobTillaart, 2023-08-26:
As the discussion continues in the private repo I close this issue

closed 2023-08-26 by RobTillaart