RobTillaart/MCP23S17 #34: MCP.read(pin) always returns 1

Obili69, 2024-01-31:
Hello Thanks for the great library
I have a problem when using all read functions with the MCP23S17. 
I have a switch on a dev board and pull PA 0-2 and PA 0-2 to either VCC(3.3V) or GND.
Im using your lib example(digital read). I changed one line. SPI.begin(); to SPI.begin(12, 11, 13, 42); 
The serial output looks like this
MCP23S17_LIB_VERSION: 0.5.0
true
1
1
HWSPI: 1
TEST read1(pin)
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 
TEST read1(pin)
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 
TEST read1(pin)

RobTillaart, 2024-01-31:
thanks for reporting this issue, as I am rather busy I have little time to investigate.
might take several days

Obili69, 2024-01-31:
No problem. Im currently also investigating the issue.

Obili69, 2024-02-02:
Some answers :)
* im using pin 42 as CS 
* Yes i had now im only getting 0 even if high  im not sure why it flipped
* No its just a switch that changes from VCC to GND

CODE:
//
//    FILE: MCP23S17_digitalRead.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: test MCP23S17 library
//     URL: https://github.com/RobTillaart/MCP23S17

#include "MCP23S17.h"

MCP23S17 MCP(42);
int rv = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.print("MCP23S17_LIB_VERSION: ");
  Serial.println(MCP23S17_LIB_VERSION);
  delay(100);

  SPI.begin(12, 11, 13, 42);

  rv = MCP.begin();
  Serial.println(rv ? "true" : "false");

  rv = MCP.pinMode8(0, 0xFF);
  Serial.println(rv);
  rv = MCP.pinMode8(1, 0x00);
  Serial.println(rv);

  Serial.print("HWSPI: ");
  Serial.println(MCP.usesHWSPI());

  Serial.println("TEST read1(pin)");
  for (int pin = 0; pin < 8; pin++)
  {
    int val = MCP.read1(pin);
    Serial.print(val);
    Serial.print(' ');
    delay(100);
  }
  Serial.println();
  MCP.write8(1, 0xFF);
}

void loop()
{
  delay(1000);
  Serial.println("TEST read1(pin)");
  for (int pin = 0; pin < 8; pin++)
  {
    int val = MCP.read1(pin);
    Serial.print(val);
    Serial.print(' ');
    delay(100);
  }
  Serial.println();

}

//  -- END OF FILE --

RobTillaart, 2024-02-02:
Could you run a test with software Spi ?
You should be able to use the same pins.

MCP23S17 MCP(select, datain, dataout, clock)

A thought popped up
Could it be that datain and dataout were swapped?

RobTillaart, 2024-02-02:
Do you use this one - https://www.mikroe.com/expand-2-click

Obili69, 2024-02-02:
I got the Mirkoe-951 
As far as i saw thats the SPI version. What i think is wierd is that the Outputs are working

Obili69, 2024-02-02:
News 
I found a second click board and i tried that. It just works now. Thanks A lot anyway. Hope you keep up the great work. I donated you a bit. Sorry for the time waste.

closed 2024-02-02 by Obili69

RobTillaart, 2024-02-02:
> Sorry for the time waste.

If you learn something, time is never wasted!
Was it a different type of board, or is the first one a faulty one? just to understand the root cause.

> Hope you keep up the great work, I donated you a bit. 

Will do so, thank you, much appreciated!

Obili69, 2024-02-02:
The board was just faulty only the port i used as a output was working i changed the port i uses as input to output and it didnt work anymore. :(  Thanks have a nice day