RobTillaart/MCP23S17 #36: Is the hardware address correct?

NNGB-kor, 2024-02-24:
check the datasheet of the 'microchip' company, the hardware address will be as shown in the image.
[image]

The hardware address is 0x00 0x01... Is this correct?
0x20 0x21 0x22... is not correct?

RobTillaart, 2024-02-24:
Thanks for the question.
Will investigate coming week

RobTillaart, 2024-02-24:
@NNGB-kor 

Had a quick look as the library just works.

The 0x2.. is handled low level in MCP23S17.cpp around line 590 by the two masks.
the 0x4_ part is the shifted 0x2_ part of the address. 

```cpp
////////////////////////////////////////////////////
//
//  PRIVATE
//

//  low level read / write masks
#define MCP23S17_WRITE_REG    0x40
#define MCP23S17_READ_REG     0x41
```

- https://github.com/RobTillaart/MCP23S17/blob/master/MCP23S17.cpp#L594

Does this solve your question?

RobTillaart, 2024-03-02:
Assuming this solved the problem, I close this issue.
Feel free to reopen if needed.

closed 2024-03-02 by RobTillaart