RobTillaart/MCP23S17 #47: enableInterrupt Pin Parameter Different from other Single Pin Interface Functions

GlibSkunk, 2024-08-02:
I presume this is not intended behavior, so I thought I'd bring this to your attention. I ran into an issue today trying to set up interrupts on specific pins using the single pin interface (rather than all the 16 pin functions). The final distinction I noticed is that this section of code (or a similar construction) appears in all single pin interface functions except for enableInterrupt().

```
uint8_t dataDirectionRegister = MCP23x17_DDR_A;
  if (pin > 7)
  {
    dataDirectionRegister = MCP23x17_DDR_B;
    pin -= 8;
  }
```

The upshot is that to set up an interrupt on GPA0, I have to use:
```
MUX.pinMode1(0,INPUT_PULLUP);
MUX.enableInterrupt(8,CHANGE);
```

This is not very intuitive and I presume all that would need to be modified within enableInterrupt() is adding the section that translates 0..15 into pins assuming 0..7 are within Port A and 8..15 are within Port B. If you agree that this should be done, I can make a pull request (my second ever, so excuse me if it takes a couple tries :) )

RobTillaart, 2024-08-02:
Thanks for the issue, 
As I am rather busy I will look into this later this month (or when time permits)

RobTillaart, 2024-11-19:
@GlibSkunk 
Finally had some time to have a first look, apologies for the too long delay.

(enable interrupt context)
Your code works with 8 bit registers where mine works on16 bit register. 
And that is a different pattern than for the other single bit functions.

Side effect should be that the indices do not match.

No time to fix it completely (including testing) yet but I will try understand the proposed code today.

closed 2024-11-19 by RobTillaart

RobTillaart, 2024-11-19:
@GlibSkunk 

Created a develop branch for version 0.6.0 which includes your fix, updated readme, updated version.
Please verify if this develop branch works if you have time. 
Thanks

PR #48 (2024-11-19): Bring enableInterrupt()/disableInterrupt() in line with other single pin interfaces
Resolves #47

I tried to keep the spirit of the code as seen in the pinMode1 function. Also happened to hit some of the enhancements noted in the comments before the enableInterrupt function.

I don't have a very comprehensive testing setup but I did utilize the changes on the own board to set up interrupts and it behaved correctly.
commit ee545d9e (2024-11-22): fix #47, enable interrupts. (#49)

- fix #47, interrupt handling. Kudos to GlibSkunk!
- update readme.md
- minor edits.