RobTillaart/MCP23S17 #50: Addressing the 16 bit registers.

RobTillaart, 2024-11-19:
extracted from - https://github.com/RobTillaart/MCP23S17/pull/48

Investigate if it is worth breaking the interface.

----

So in general I believe the the 16 functions are a bit conceptually out of whack with the single pin versions. However I see that this has to do with what arises naturally out of the MCP23S17's register map rather than any Library issue.

The natural 0..15 (used in single pin functions) is:
GPA0, GPA1 ... GPA7, GPB0, GPB1, ... GPB7

The 0..15 from the datasheet (used in 16 pin functions) is:
GPB0, GPB1 ... GPB7, GPA0, GPA1, ... GPA7

I think the only way to get consistency would be to swap the first and second byte so that you could use reg[3] to access GPA4, matching the "natural" way (for me at least). I don't think using [15..0] would have the desired effect.

However, I'm honestly not sure that's worth doing regardless. The 16 mode as implemented matches the datasheet so I'd say it's best to leave it alone and let people choose to either use the 16 bit "full" interface or the single bit at a time interface. The only reason for my difficulty was because the enableInterrupt function appeared to match the pinMode1 function but didn't.

----

Alternatively, I had another thought of how to achieve consistency (again I don't know that it's worth upending the library).

This section of code in begin() exists to ensure the sequential mode is active:

```
//  disable address increment - datasheet P20
//  note that address increment must be enabled for readReg16() and writeReg16() to work.
//    SEQOP: Sequential Operation mode bit
//    1 = Sequential operation disabled, address pointer does not increment.
//    0 = Sequential operation enabled, address pointer increments.
uint8_t reg = readReg(MCP23x17_IOCR);
if (reg & MCP23x17_IOCR_SEQOP)  //  check if already zero
{
  reg &= ~MCP23x17_IOCR_SEQOP;  //  clear SEQOP bit for sequential read/write
  if (! writeReg(MCP23x17_IOCR, reg)) return false;
}
```

The phrase "note that address increment must be enabled for readReg16() and writeReg16() to work" is actually not quite accurate. On page 13 of the datasheet - 

> A special mode (Byte mode with IOCON.BANK = 0)
causes the address pointer to toggle between
associated A/B register pairs. For example, if the BANK
bit is cleared and the Address Pointer is initially set to
address 12h (GPIOA) or 13h (GPIOB), the pointer will
toggle between GPIOA and GPIOB. Note that the
Address Pointer can initially point to either address in
the register pair.

Because all read16 and write16 work on A/B pairs (that I've seen at least), even with sequential mode off, they would work and bounce between the registers. So my proposal is: turn off sequential mode (IOCON.SEQOP=1) and target the B port for all read16/write16. That way the upper byte is associated with the B port and the following lower byte is associated with the A port.

Obviously this would be a breaking change and warrants more consideration, but I think it would have the desired outcome.

Andy2No, 2025-03-16:
Wouldn't it be better to provide an option to reverse the byte ordering for the single pin functions?  It seems A was meant to be the MSB and B was meant to be the LSB, since that's how the 16 bit operations work.  The "natural" ordering seems to have been the result of choosing B as the MSB instead.

So, to keep compatibility, you might keep the same default behavior but offer the option to swap the byte ordering in the single pin operations instead.

RobTillaart, 2025-03-16:
@Andy2No 

Thanks for your question. Breaking the existing interface was the reason not to implement this, as most people use either the single pin or the 16 pin interface. Still it is a good point. 

Last month I got the same question for the MCP23017 - https://github.com/RobTillaart/MCP23017_RT/issues/44

The solution implemented there was to be able to set a flag to have all 16 bit functions swap their MSB / LSB.

I propose to implement that flag in the MCP23S17 too.

RobTillaart, 2025-03-16:
@Andy2No 

Will label it as bug to give it more priority. 
As the code is verified in the MCP23017 I expect no problems syncing this functionality into the MCP23S17

RobTillaart, 2025-03-17:
@Andy2No 

Just checked and it was already added to MCP23S17 repo.

See .h file

```cpp
  //       Fix #50, reverse the byte order of the 16 bit API.
  //       reverse == false ==> backwards compatible (default)
  //       reverse == true ==> swaps the A and B byte to be more intuitive.
  void     reverse16ByteOrder(bool reverse = false); 
```

closed 2025-03-17 by RobTillaart

Andy2No, 2025-03-17:
Okay.  Thanks.  I was actually suggesting swapping the byte order used in single pin operations instead, because that removes the need to do a byte swap on the data fed to a 16 pin operation but still makes them consistent.

E.g.:

```
bool MCP23S17::write1(uint8_t pin, uint8_t value)
{
  if (pin > 15)
  {
    _error = MCP23S17_PIN_ERROR;
    return false;
  }
  uint8_t IOR = MCP23x17_GPIO_A;
  if (pin > 7)
  {
    IOR = MCP23x17_GPIO_B;
    pin -= 8;
  }

// This could be changed to match 16 bit operations without requiring any extra execution steps (leaving out the lines needed to make this behaviour switchable):

bool MCP23S17::write1(uint8_t pin, uint8_t value)
{
  if (pin > 15)
  {
    _error = MCP23S17_PIN_ERROR;
    return false;
  }
  uint8_t IOR = MCP23x17_GPIO_B;
  if (pin > 7)
  {
    IOR = MCP23x17_GPIO_A;
    pin -= 8;
  }
```

I was under the impression that was the manufacturers intended addressing scheme, but I guess it doesn't really matter either way, and the byte swap is still quite quick.

RobTillaart, 2025-03-17:
@Andy2No 
Would work to, however be a breaking change for installed projects. 
The way now solved (which is a band aid I agree) does not break them and allows other project to be more consistent.

As I mentioned in post nr 1 the datasheet and the natural order conflict a bit. 

> the byte swap is still quite quick.

The swapping of the bytes could be a bit faster by using an union of the 16 bit value and two 8 bit MSB LSB.
Then the compiler can access all data directly.
If performance is a problem, first increase the SPI speed as clocking the bits takes most time.

commit 5f1f11f1 (2025-02-18): Fix #50
commit 6eb267a8 (2025-02-18): Fix #50
commit 06fdb352 (2025-02-18): Fix #50,  byte order 16 bits API (#53)

- fix #50, add function to reverse the byte order of the 16 bit API.
- update readme.md