RobTillaart/MCP23S17 #54: Compile error with Portenta H7

verityworks, 2025-04-15:
Im porting my code from an Arduino Portenta C33 to an Arduino Portenta H7
my .ino file has this constructor, since im using SPI1:

MCP23S17 MCP(PB_9,&SPI1);

When trying to compile your library, I have this error:

C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.2.4\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h747xx.h:2650:55: error: lvalue required as unary '&' operand
 #define SPI1                ((SPI_TypeDef *) SPI1_BASE)
                                                       ^
C:\Users\xxx\AppData\Local\Temp\.arduinoIDE-unsaved2025315-32116-6ozhyw.x1vfa\MCP23S17_digitalRead\MCP23S17_digitalRead.ino:11:20: note: in expansion of macro 'SPI1'
 MCP23S17 MCP(PB_9,&SPI1);
                    ^
exit status 1

RobTillaart, 2025-04-15:
Thanks for reporting the issue.
As I do not have Portenta H7 experience I can only guess what is happening right now.
Furthermore I have some other tasks I have to fix asap so it might take a few days before I can dive into it.

RobTillaart, 2025-04-15:
@verityworks 

Spend an hour or so investigating and the cause is that the **pins_arduino.h:145:23** file defines only 1 SPI port.

```
 #define SPI_HOWMANY   1
```

So SPI1 is not defined in SPI.h

```
#if !defined(ARDUINO_AS_MBED_LIBRARY)

#if SPI_HOWMANY > 0
extern arduino::MbedSPI SPI;
#endif

#if SPI_HOWMANY > 1
#ifdef SPI1
#undef SPI1
#endif
extern arduino::MbedSPI SPI1; 
#endif
```

And then it ends with a second definition of SPI1 here which is of an incompatible type as far as I understand.
The error you got.

...packages\arduino\hardware\mbed_portenta\4.1.5\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h747xx.h:2650:55

### Conclusion

in short, SPI1 is NOT supported by Arduino for the Portenta H7.

RobTillaart, 2025-04-15:
As this is not a library problem you can close the issue.

verityworks, 2025-04-16:
I see, its indeed some kind of bug with the Portenta H7 libs I think, everything worked fine with the Portenta C33, which should be pin compatible to the Portenta HAT it uses.
I was also working with the MCP3304, which is using the same MOSI,MISO,CLK lines but different CS pin. And I got that to work using raw SPI commands (SPI.begintransaction, SPI.transfer.. etc)
So, i know electricaly speaking, its connected right.  I just dont see myself getting data out of that much more complex MCP23S17 using raw SPI commands.
I know its not your library, but do you have any idea what could be wrong with the Portenta Libs?

RobTillaart, 2025-04-16:
Why don't you use the default SPI, as you only need a select pin ?

```cpp
MCP23S17 MCP(PB_9);
```

Or does this give other errors.

RobTillaart, 2025-04-16:
> I was also working with the MCP3304, which is using the same MOSI,MISO,CLK lines but different CS pin. And I got that to work using raw SPI commands (SPI.begintransaction, SPI.transfer.. etc)

interesting chip, added to my "to investigate" list, it is probably similar to the 3208 ADC

RobTillaart, 2025-04-16:
FYI, the MCP3304 looks very much like the MCP_ADC so I created an experimental library.
- https://github.com/RobTillaart/MCP330X  (under develop branch now, release tomorrow.)

If you have time, you may give it a try.

verityworks, 2025-04-17:
> Why don't you use the default SPI, as you only need a select pin ?
> 
> MCP23S17 MCP(PB_9);
> Or does this give other errors.

I dont see any inputs that way. Its weird.  Im gonna connect some leds to the CS pins and see if they are actually working (C33 vs H7)
Our IO board that connects to the portenta HAT has 2x MCP23S17 and one MCP3304.

Looking forward to your MCP3304 lib.

verityworks, 2025-04-17:
I have found the solution why I have a problem.

I use PB_9, PI_2, PI_3, as pinnames to set the chipselect pins. These defines go to 25, 130, 130.
However, these are not the pinnames that actually drive an output.

This function in the mbed "wiring_digital.cpp" has 2 versions. One for the "PB_9" type pinnames, and then another for regular uint types. For some reason the PinNameToIndex sets the wrong pin number.

When I debugged the code
PB_9 gave idx 51
PI_2 gave idx 156
PI_3 gave idx 157

when I set those numbers as input to your MCP function, it all started working.

Not sure where I post this info so Arduino team can look at this.

```
void pinMode(PinName pin, PinMode mode)
{
  pin_size_t idx = PinNameToIndex(pin);
  Serial.print("IDX:");
  Serial.println(idx);
  if (idx != NOT_A_PIN) {
    pinMode(idx, mode);
  } else {
    switch (mode) {
      case INPUT:
        mbed::DigitalIn(pin, PullNone);
        break;
      case OUTPUT:
        mbed::DigitalOut(pin, 0);
        break;
      case INPUT_PULLUP:
        mbed::DigitalIn(pin, PullUp);
        break;
      case INPUT_PULLDOWN:
      default:
        mbed::DigitalIn(pin, PullDown);
        break;
    }
  }
}

void pinMode(pin_size_t pin, PinMode mode)
{
  if (pin >= PINS_COUNT) {
    return;
  }
  mbed::DigitalInOut* gpio = digitalPinToGpio(pin);
  if (gpio != NULL) {
    delete gpio;
  }
  gpio = new mbed::DigitalInOut(digitalPinToPinName(pin));
  digitalPinToGpio(pin) = gpio;

  switch (mode) {
    case INPUT:
      gpio->input();
      gpio->mode(PullNone);
      break;
    case OUTPUT:
      gpio->output();
      break;
    case INPUT_PULLUP:
      gpio->input();
      gpio->mode(PullUp);
      break;
    case INPUT_PULLDOWN:
    default:
      gpio->input();
      gpio->mode(PullDown);
      break;
  }
}
```

RobTillaart, 2025-04-17:
> Looking forward to your MCP3304 lib.

will be released in 30 minutes if merge doesn't play tricks with me :)

RobTillaart, 2025-04-17:
https://github.com/RobTillaart/MCP330X

Released, to be expected in the library manager within 24 hours, or download directly from GitHub.

RobTillaart, 2025-04-17:
@verityworks 

As the problems seem to be solved you can close the issue.

closed 2025-04-18 by RobTillaart