RobTillaart/MCP23S17 #55: Invalid definition of Constructor

crocodileInside, 2025-04-25:
The definition of the following constructor in MCP23S17.cpp is invalid:

`
//  HARDWARE SPI
MCP23S17::MCP23S17(int select, __SPI_CLASS__ * spi)
{
  MCP23S17(select, 0x00, spi);
}
`

This just creates a new instance of the MCP23S17 object _inside_ the constructor, but discards it afterwards. Therefore the original MCP23S17 instance is still not initialized, despite calling the constructor. This causes a nullptr exception afterwards when you try to use the library, since `_mySPI` was never set correctly. 

The constructor code should look like this: 

`
//  HARDWARE SPI
MCP23S17::MCP23S17(int select, __SPI_CLASS__ * spi)
  : MCP23S17(select, 0x00, spi)
{
}
`

RobTillaart, 2025-04-25:
Thanks for reporting the issue,
Labelled it as bug to give it priority.

RobTillaart, 2025-04-25:
@crocodileInside 
wrote a minimal class to verify and (1) bug confirmed and (2) solution confirmed.

Will fix asap - might take an hour or so.

RobTillaart, 2025-04-25:
Created develop branch and PR, just made a straightforward constructor. 
build is running

RobTillaart, 2025-04-25:
@crocodileInside 
Created linked issue in the MCP23S08 library as it has same (faulty) implementation.

RobTillaart, 2025-04-25:
@crocodileInside 

Thanks for reporting the issue, appreciated!

commit 4cb96b06 (2025-04-25): fix #55, hardware constructor
commit 8fbef270 (2025-04-25): fix #55, hardware constructor (#56)

- fix #55, hardware constructor, kudos to crocodileInside
- update readme.md
closed 2025-04-25 by RobTillaart